Theta-Limited/OpenAthenaAndroid #67: Mavic mini 1 seems to do not have geolocalization data in exif...

RomainB91, 2023-05-02:
Hi,

Thanks a lot for OpenAthena for Android !

I cannot work with Open Athena for Android and my mavic mini 1.

I do not understand : It seems that my Mavic mini 1 seems to do not have geolocalization data in exif under irfanview.
But when i look at the pic in my android phone... it show me some geolocalization data in google map !

Please could you help me, i do not understand.

mkrupczak3, 2023-05-04:
Hi @RomainB91, 

Are you having trouble using with OpenAthena for Android with your drone images? If so, feel free to send some in an email ([matthew.krupczak@theta.limited](mailto:matthew.krupczak@theta.limited)) and I'd be happy to take a look.

In general, photo metadata for drone images is either stored in either [EXIF](https://exiftool.org/TagNames/EXIF.html) metadata, Adobe [XMP](https://exiftool.org/TagNames/XMP.html) metadata, or both. If you send images via a messaging app or social media, this data (especially location tags) is sometimes automatically removed to protect user privacy. This may explain why you can see the location data on your phone but not on your computer. You can use a free tool called [ExifTool](https://exiftool.org/) to check if the data has been removed or zeroed out after your transfer.

Your Mavic Mini 1 has camera model `FC7203`. I've tried a [few images from online](https://commons.wikimedia.org/wiki/Category:Taken_with_DJI_FC7203), and believe you could expect the following:

[image]

Latitude, Longitude, and Altitude will be present in the EXIF tags, but not XMP tags. Camera gimbal yaw and pitch are stored within XMP tags. This should be sufficient to work with OpenAthena.

Some older DJI drones incorrectly use their relative altitude from their start point as their "absolute altitude". If so, the data will be incompatible with OpenAthena. Sometimes this can be fixed with a firmware update, but this is not guaranteed.

If the data has been removed by your messaging app or other software, the Exif.GPSInfo tags will be zeroed out like this:

[image]

mkrupczak3, 2023-05-20:
I'm keeping this issue open until I can do more testing with the Mavic Mini

RomainB91, 2023-05-21:
Hi, thanks a lot.

If i could help you providing more pictures from the mini (i have a lot of
new ones :-)) or anything else, do not hesitate to ask me.

Best regards,

Romain

mkrupczak3, 2023-06-10:
Hi @RomainB91,

Wanted to let you know that v0.16.0 is out, with vastly improved accuracy

This may reduce some of the problems you encountered earlier

RomainB91, 2023-06-10:
Hi,

Thanks for the update :)

Have a nice weekend !

Best regards,

Romain

closed 2024-04-01 by mkrupczak3