Theta-Limited/OpenAthenaAndroid #96: Fastlane issues

IzzySoft, 2023-10-07:
`full_description.txt` and `short_description.txt` for `zh-CN` are empty (0 byte) files which causes some issues: the "empty" description might overwrite the fallback `en-US` one causing the app to show up without any description for that locale – or as in my case, it throws an "failed download: empty file" error on each update.

So could you please either fill those files with content, or remove them entirely? Thanks in advance!

Oh, and btw: you might wish to [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) to accompany the lonely PlayStore one, linking it to `https://apt.izzysoft.de/packages/com.openathena/` :wink:

mkrupczak3, 2023-10-10:
Hi @IzzySoft 

Thanks for your interest in OpenAthena!

Theta's business is predicated on the successful delivery of trustable, signed, open source software products. Availability on F-Droid and 3rd party repositories like yours is a high priority for our firm.

I haven't been keeping the `metadata` tags up to date since our initial submission to F-Droid was never serviced. Please allow a few days for me to update these tags. Our app contains localization for French, German, Polish, Japanese, Korean, Russian, Ukrainian, and Simplified Chinese in addition to English, so our metadata tags will target the same languages.

I believe F-Droid keys off of GitHub version tags, so I will make a minor version release once the metadata is updated.

Please let me know if you have any issues with this repo thereafter. I can make further edits and even review pull requests to make sure our repo is usable.

You are free to distribute OpenAthena for Android as you wish, on my end however I will need to review IzzySoft's method of distribution before adding a badge to the README.md. We only link to distributions which provide binaries signed with my private key.

If you have any other questions or concerns, feel free to reach out to me at matthew@theta.limited

-Matthew

IzzySoft, 2023-10-10:
Hi @mkrupczak3 and thanks for your response! Yes, I indeed noticed that in the logs of my updater (my repo, too, pulls fastlane from here; other than F-Droid it doesn't pull directly at tag level but picks up what is provided at `HEAD` at the time the update is fetched. My updater doesn't use `git pull` but rather works via Github's (and GitLab's and Codeberg's/Gitea's/ForgeIO's) API).

> since our initial submission to F-Droid was never serviced

Sad to read this! Did it get stuck, or was it closed/rejected? I don't see any show-stoppers in my scanner's report, but F-Droid runs additional checks and has harder requirements than my repo.

> Availability on F-Droid and 3rd party repositories like yours is a high priority for our firm.

Then you'll probably also love to add the above mentioned badge to accompany the currently lonely PlayStore badge :smiley:

> We only link to distributions which provide binaries signed with my private key.

So go right ahead! My updater picks the APKs you provide at your releases here. And for security reasons makes sure the signing key wasn't changed (e.g. by some malicious actor having somehow gained access to an app's repo), so you can check your certificate and compare its SHA256  to the one the app is pinned to here: `4faa6dab09f9d15ac204ea8b8c056ccaf8c52993007cd643001a3d3497c7944b`.

> I believe F-Droid keys off of GitHub version tags, so I will make a minor version release once the metadata is updated.

That indeed would automatically trigger an update here within 24h. Other than F-Droid, I can also trigger a manual reload of fastlane data separately – but an update would bring the app "to the front page" again causing more people to take a look and see your refreshed descriptions, screenshots etc.

> Please let me know if you have any issues with this repo thereafter. I can make further edits and even review pull requests to make sure our repo is usable.

Thanks, I'll gladly take another look then – and if you wish even a closer one to be able to give some hints on possible improvements. Until then, be welcome to take a look at:

* my [Fastlane Cheat Sheet](https://gitlab.com/-/snippets/1895688)
* [my Wiki on Fastlane formattings](https://gitlab.com/IzzyOnDroid/repo/-/wikis/Fastlane)

As for reviewing the process: you can find most details (including the code used) at the [corresponding GitLab repo](https://gitlab.com/IzzyOnDroid/repo/) – and feel free to ask if something was left unclear. TL;DR: APKs are pulled from releases, metadata from Fastlane if they exist. APKs are checked with my library scanner (which is also used by F-Droid) and against VirusTotal, with the results being made transparent to users on the WebIF (there's unfortunately no way yet to inject it to the data shown by the F-Droid Android clients). The updater usually runs once a day for the regular apps; for apps which are rarely updated (e.g. haven't seen a commit for a year) there's a separate monthly run.

mkrupczak3, 2023-10-11:
Ok, I've updated the en-US changelogs in 3de9b1b925b357e4152e35086c220ca00128f4db and added localized descriptions for each language.

I need to add some screenshots in a few languages, then I will push out a tag v0.19.2

I'll attempt to get a submission to F-Droid shortly after.

Will add the IzzySoft badge in first commit after v0.19.2 tag

mkrupczak3, 2023-10-11:
Out of curiosity, how did you hear about OpenAthena for Android?

We've only recently started marketing efforts, feedback could be helpful

mkrupczak3, 2023-10-11:
@IzzySoft I updated metadata for all locales in 8e6efce43f813f83e3fe3add2d329c588c91b7a0

A new release, v0.19.2 with version code 27 is now live:
https://github.com/Theta-Limited/OpenAthenaAndroid/releases

I was able to build the app using the `fdroidserver` container. I have submitted a Merge Request for inclusion into FDroid here:
https://gitlab.com/fdroid/fdroiddata/-/merge_requests/13839

The APKSigningBlock and APKSigningBlockOffset files for reproducible builds are available from the MR here:
https://gitlab.com/mkrupczak/fdroiddata/-/tree/com.openathena/metadata/com.openathena/signatures/27

It should be ready for inclusion with IzzySoft given these changes.

Let me know if you can build the app, and when your publication is live I can add the IzzySoft badge to this repo.

mkrupczak3, 2023-10-11:
The app contains two (non-code) binary data files:
`app/src/main/resources/EGM96complete.dat.gz`

`app/src/main/assets/EGM96complete.bin`

...which cause the default FDroid build pipeline to fail (due to it misinterpreting them as non-free code binaries).

The project is buildable with the `fdroid build --skip-scan` command.  I made a more detailed writeup of what these files are and their provenance here:
https://gitlab.com/fdroid/fdroiddata/-/merge_requests/13839#build-errors-due-to-fdroid-scan-misinterpreting-the-purpose-of-targz-and-bin-data-files

IzzySoft, 2023-10-11:
> Out of curiosity, how did you hear about OpenAthena for Android?

One of my crawlers found it back in February :wink: 

[image]

> It should be ready for inclusion with IzzySoft given these changes.

"Ready for update" you mean :stuck_out_tongue_closed_eyes:

[image]

OK, let's go:

```
$ iod repo get com.openathena
com.openathena: looking for 'https://api.github.com/repos/Theta-Limited/OpenAthenaAndroid/releases'
com.openathena: checking tag 'v0.19.2'
com.openathena: lastRelNo set to '0.19.2', checking for files
com.openathena: Upstream file date (2023-10-11 05:05) is newer than ours (2023-10-07 19:45).
com.openathena: returning ['0.19.2','https://github.com/Theta-Limited/OpenAthenaAndroid/releases/download/v0.19.2/OpenAthena_for_Android_v0.19.2.apk',1696993500]
com.openathena: 0.19.1/0.19.2, https://github.com/Theta-Limited/OpenAthenaAndroid/releases: https://github.com/Theta-Limited/OpenAthenaAndroid/releases/download/v0.19.2/OpenAthena_for_Android_v0.19.2.apk
- Grabbing update for com.openathena: OK
- Checking 'repo/com.openathena_27.apk' for libraries and malware …
com.openathena: check if repo contains FUNDING.yml
com.openathena: looking for 'https://api.github.com/repos/Theta-Limited/OpenAthenaAndroid/contents/.github'
com.openathena: looking for 'https://api.github.com/repos/Theta-Limited/OpenAthenaAndroid/contents/'
com.openathena: no FUNDING.yml detected.
com.openathena: checking Fastlane for per-release changelogs
com.openathena: fetched 'https://github.com/Theta-Limited/OpenAthenaAndroid/raw/master/metadata/en-US/changelogs/27.txt'
com.openathena: calling 'getFastlaneMeta(github,[host:github.com,owner:Theta-Limited,repo:OpenAthenaAndroid,path:/metadata])'
com.openathena: FastlaneFeatures shortdesc,fulldescMD,changelogs,icon,screenshotsJPG
…
```

(and 2 more pages with the fastlane details) – thanks, all pulled fine, and no errors :smiley: Update will go live with the next sync around 6 pm UTC. Btw: you can use "simple HTML" in your `full_description.txt` if you want for e.g.

```markdown
Please see <a href='https://OpenAthena.com/'>OpenAthena.com</a> for more information
```

[image]

(thanks for including changelogs with fastlane – they are now integrated as well as you can see)

> The app contains two (non-code) binary data files:

Ah, I see. So that tripped the scanner and resulted in an error. As it's easy to explain (and not a real issue), those files can be "whitelisted" in the build recipe at F-Droid using `scanignore`, so please mention it with your RFP. Those files should be no show-stoppers.

mkrupczak3, 2023-10-12:
IzzySoft badge added in 0613fedc875204fc5cba83c9594f0679528eaaba and f1091785c9ba8af5d39eb420a8ec7c333990a900. Marking this issue as closed

IzzySoft, 2023-10-12:
Great, thanks! I see that your inclusion with F-Droid.org is progressing well, congrats! And it seems you even established reproducible builds (though at the moment it's not clear whether that succeeded, as you used the method that requires you to submit the signing blocks per MR – which also means that IMHO Auto-Update cannot work as designed), cool! So once your app shows up there, do you want me to keep it in my repo? As by that method, updates to F-Droid would (if I understand correctly) always require manual work: a MR with the signing blocks that must be approved and merged, so updates will take significantly longer. I wonder why you've chosen that path instead of using

```yaml
Binaries: https://github.com/Theta-Limited/OpenAthenaAndroid/releases/download/%v/OpenAthena_for_Android_%v.apk
```

which could be handled automatically (`%v` is automatically replaced by the corresponding `versionName`, the resulting APK downloaded & compared to the one F-Droid built – and if they match (ignoring the signature) the APK is shipped with your signature).

mkrupczak3, 2023-10-12:
Hi,

Thanks for the feedback, It's been very helpful for my submission process to FDroid

I'm going to implement your suggestion to use the `Binaries:` field rather than importing signed block manually.

Fixing up another commit for that MR now...

mkrupczak3, 2023-10-12:
I've added a couple of new commits to fix syntax and implement the `Binaries:` field to allow for autoupdates.

Watching the build process now

mkrupczak3, 2023-10-12:
Reproducible builds seem to be failing, not sure why:
https://gitlab.com/fdroid/fdroiddata/-/merge_requests/13839#note_1601179995

mkrupczak3, 2023-10-12:
I'm AFK for the rest of this evening, will try again tomorrow

IzzySoft, 2023-10-13:
Thanks! Answered "over there" to keep things together.

mkrupczak3, 2023-10-15:
[Merge request with F-Droid was accepted](https://gitlab.com/fdroid/fdroiddata/-/merge_requests/13839#build-errors-due-to-fdroid-scan-misinterpreting-the-purpose-of-targz-and-bin-data-files), OpenAthena for Android may be showing up there soon.

Marking this issue as closed

IzzySoft, 2023-10-15:
@mkrupczak3 Congrats! May I, in this context, repeat my question [from above](https://github.com/Theta-Limited/OpenAthenaAndroid/issues/96#issuecomment-1760200207): Do you want me to keep OpenAthena in my repo once it is available at F-Droid.org? Usually I remove apps at that stage (after a decent overlap of 10..14 days). But on request by their authors, especially with reproducible builds being established, keeping it here is an option if needed.

mkrupczak3, 2023-10-16:
@IzzySoft I don't see any problem with keeping it available on IzzySoft, and in fact it might improve overall availability of the app

I'll put an F-Droid badge with the others when the app appears there

IzzySoft, 2023-10-16:
As I wrote, it's technically possible if requested. Say you want me to keep it, and I mark it accordingly (else it surfaces for removal once it shows up at F-Droid).

mkrupczak3, 2023-10-17:
Please keep it

IzzySoft, 2023-10-17:
Marked:

[image]

Btw, the German description is a bit more detailed, in case you want to adopt some parts (I can send you both for fastlane):

[image]

mkrupczak3, 2023-10-17:
Sure, I can review that either as a pull request or just the raw text if you prefer.

The app hasn't shown up on F-Droid yet, I'm thinking might be because I had a changelog present for version code 27 but not 28. I just pushed out a bug hotfix with version code 29, with any luck it will show up on F-Droid soon

mkrupczak3, 2023-10-17:
Maybe I should add German localization soon, as the 9th supported language? @IzzySoft Would you be interested in helping with a QA review once a localization is ready?

IzzySoft, 2023-10-18:
> The app hasn't shown up on F-Droid yet,

Index publish is just pending. I didn't look which apps where built this cycle, but we should know pretty soon™ – as soon as signing is done and the index published.

> Maybe I should add German localization soon

Shall I make a PR with what I have at my end, to combine both your comments? Will do. I'll also resolve the "circular reference" then (fulldesc sends readers to your homepage for more information – but all currently found there is basically a link to the Github repo were said details have to be searched for; I've linked that directly. Also, my descriptions are HTML "compressed" to one line; reason for the latter is that otherwise F-Droid replaces each `\n` by `<br>` causing "ugly holes".

There you go :smiley:

IzzySoft, 2024-08-18:
PS: monthly quality checker here just complained:

    ! com.openathena/en-US/changelogs/41.txt has 534 chars

Fastlane limit there is 500 chars, everything behind that will simply be cut off, even mid-word. When taking a look at that changelog if I could bring it back into limits (I succeeded heavily there) I saw you're using HTML there. Well, that's not permitted – only plain-text. Hence the results look pretty ugly. Please use plain-text (and not more than 500 chars) there with future releases. Thanks!

commit 6b966378 (2024-08-23): fix #96 fastlane metadata quality issue
closed 2024-08-23 by mkrupczak3