Theta-Limited/OpenAthenaAndroid #104: The effect of aircraft roll value

kubbo, 2024-01-10:
The code only references the roll value of the camera. If the roll value of the aircraft is not 0, will there be any issues with the result?

kubbo, 2024-01-12:
I'm just adding, why is there no roll value for the camera in resolveTarget(double lat, double lon, double alt, double azimuth, double theta)

mkrupczak3, 2024-01-12:
We handle roll value when the metadata is first extracted from the image in MetadataExtractor.java in getRayAnglesFromImgPixel:
https://github.com/Theta-Limited/OpenAthenaAndroid/blob/056032ebefeb91805eb4803d7b881f6a4c8eac9d/app/src/main/java/com/openathena/MetadataExtractor.java#L800

That function calls correctRayAnglesForRoll, using the roll value from the image metadata.

The correctRayAnglesForRoll function applies the camera's roll to the azimuth and downward angle, providing new values corrected for the camera's roll:

https://github.com/Theta-Limited/OpenAthenaAndroid/blob/056032ebefeb91805eb4803d7b881f6a4c8eac9d/app/src/main/java/com/openathena/MetadataExtractor.java#L825

Because we handle roll in the MetadataExtractor, we don't need to do anything with it in resolveTarget. resolveTarget assumes that the azimuth and theta angle have already been corrected for roll. 

That might be a bit confusing, hope it makes sense

closed 2024-01-13 by mkrupczak3