Theta-Limited/OpenAthenaAndroid #150: Check for presence of DEM API Key on application launch

mkrupczak3, 2024-07-09:
...and direct the User to obtain and input an API Key if not present

closed 2024-07-09 by mkrupczak3

rdkgit, 2024-07-10:
This entire step should NOT be needed.  We obtained an OT key for bundling with Android app much like iOS app.

mkrupczak3, 2024-07-10:
> This entire step should NOT be needed. We obtained an OT key for bundling with Android app much like iOS app.

If the user downloads app from GitHub or F-Droid, we will not be able to bundle a key. 

F-Droid does reproducible builds from source code and compares with the binary on our GitHub release. If we alter the build from the public source code to include a key, it would break reproducible builds.

There is a non-trivial community of users who prefer F-Droid over Google Play due to privacy and security concerns. We need to support these users as much as possible

rdkgit, 2024-07-10:
Ahh, makes perfect sense.  Key bundled with GooglePlay app but not source nor alt app stores.

commit b9ad6e94 (2024-07-09): fix #150 check for presence of DEM API Key on app launch