Theta-Limited/OpenAthenaAndroid #179: build issue

IzzySoft, 2024-11-08:
There seems to be a problem building the latest release:

```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileReleaseJavaWithJavac'.
> Cannot access input property 'stableSources' of task ':app:compileReleaseJavaWithJavac'. Accessing unreadable inputs or outputs is not supported. Declare the task as untracked by using Task.doNotTrackState(). See https://docs.gradle.org/8.0/userguide/incremental_build.html#disable-state-tracking for more details.
   > Failed to create MD5 hash for file '/build/repo/app/src/main/java/com/openathena/CK42_Gauss_Kr??ger_Translator.java' as it does not exist.
```

That file name has an umlaut (`ü`: `CK42_Gauss_Krüger_Translator.java`) causing issues (building on Debian bookworm here). Could you (with the next release) maybe rename that to `CK42_Gauss_Krueger_Translator.java` to avoid the umlaut?

Thanks in advance!

mkrupczak3, 2024-11-20:
Encountered a similar issue in building [OpenAthena Core](https://theta.limited/openathena-core/), where the umlaut was breaking the build on macos arm64.

Just fixed in eca82a8 and will be present in the v0.22.0 release imminently. Thanks for the report!

closed 2024-11-20 by mkrupczak3

IzzySoft, 2024-11-20:
Cool, thanks! Can you give me a ping when the new release is out? Background: I'm trying for [Reproducible Builds](https://reproducible-builds.org/) (see: [Reproducible Builds, special client support and more at IzzyOnDroid](https://android.izzysoft.de/articles/named/iod-rbs-mirrors-clients)), so I'd not only need to build myself but also have your build to compare against, built from the very same commit.

mkrupczak3, 2024-11-20:
Awesome! Glad it worked. Thanks for the heads up on the build issue

The version on GitHub is slightly different from the Google Play one; I had to remove the OpenTopography API key from the public one so as not to leak it. 

The app links to instructions for obtaining an API key though. It can also be used without the OpenTopography service if the user loads elevation model files into the app's cache directory

IzzySoft, 2024-11-20:
> Glad it worked.

Me too! And thanks for your help again as well!

As for the API keys: you know they can be extracted from the APKs as well usually? I sometimes do that to get an app RB: extract the keys, encode them base64 to not make them too obvious, and the decode at build time (e.g. `base64 -d > local.properties <<< R09PR0xFX0....`). I know that's not encrypted – but neither is it encrypted in the APK :man_shrugging: But yeah, telling people how to obtain their own for free and giving them the option to import them to the app is much better, good thinking!

mkrupczak3, 2024-11-20:
Having to obtain one manually is still a big hassle for users, especially since the process is difficult on OpenTopography's website for mobile. Not ideal but the thought is that F-Droid or GitHub users are more technically-inclined anyways.

I might do something like that obfuscation for a future release

commit eca82a8a (2024-11-20): fix #179 build issue due to non-ascii character in a class name